SunDevilRocketry/Flight-Computer-R3 #6: Add a secondary Flash module

niekky, 2025-02-14:
As a requirement, the rev3 shall have a secondary flash memory so that it can store configuration/preset data without affecting the primary flash memory that's mainly for flight data storage.

Chip requirements:
- The chip needs to be interfaced through SPI
- The chip shall be powered by 3.3V
- The chip doesn't need to have memory as big as the main flash memory. (Configuration data is small and is not stored repetitively)
- The chip should be smaller than the main flash memory in terms of size.

dmortloff, 2025-02-26:
Added 2nd Microchip Technology Flash (SST25VF512A-33-4I-SAE) instead of previous flash
Added ground to UART connector, and 4xed schematic
New Flash can be found here
https://www.digikey.com/en/products/detail/microchip-technology/SST25VF512A-33-4I-SAE/2299415 
[image]

niekky, 2025-02-26:
[image]

Looks nice but here are some minor things:
- You're using European (not US) resistor symbol, Could you change it to the US symbol? 

[image]

- 10k label should be enough instead of "10kOhm"

For UART, could you rotate and have the Earth symbols pointing downward?

closed 2025-02-28 by niekky