SunDevilRocketry/Flight-Computer-R3 #7: Add an UART terminal

niekky, 2025-02-14:
As requirements, the rev3 shall have an UART terminal port so that it can connect and wire communicate with other SDR boards. 

Requirements:
- The port shall have a 5V power rail
- The port shall only have UART TX and RX wires for communication

dmortloff, 2025-02-21:
Repurposed 440054 Schematic and footprint (previous use servo PWM connection terminals) to develop new UART port
Features 5V, TX and RX starting with labels PUART to indicate peripheral UART

[image]

dmortloff, 2025-02-26:
4xed UART connectors with additional pin, and added GND tab on bottom

[image]

closed 2025-02-28 by niekky